PilzDE/pilz_industrial_motion #25: Non-empty start states invoke error despite not being used.

martiniil, 2018-11-22:
### Commit
e8f36e522815ce74a4d96bbecc7c3e0557269b3b

### Steps to reproduce
1. Create a sequence request where the second goal has a non-empty start state.
2. Plan the sequence trajectory.

### Expected behavior
The trajectory should be generated ignoring the start state of the second request completely.

### Observed behavior
The trajectory generation fails.
> MoveItErrorCodes::START_STATE_VIOLATES_PATH_CONSTRAINTS

This is unnecessary since the start state of the second request is set to the end state of the previous trajectory. 

### Reference to test
Here is a test using an invalid start state, but the issue exists for **any** non-empty start state. 
[integrationtest_command_list_manager.cpp](pilz_industrial_motion/pilz_trajectory_generation/test/integrationtest_command_list_manager.cpp):
test case: startStateNotFirstGoal

agutenkunst, 2019-01-07:
I would not agree on the behavior you expect. 
The code documentation clearly states that

https://github.com/PilzDE/pilz_industrial_motion/blob/d526457f417a8107177b5cfa28d9260c3eaee834/pilz_trajectory_generation/include/pilz_trajectory_generation/command_list_manager.h#L58-L62

However I think that this should be part of the documentation of the public interface.

Therefore I opened https://github.com/PilzDE/pilz_industrial_motion/pull/47

martiniil, 2019-01-08:
Ok, I missed that part of the documentation.

But I think the error code is not well-suited here. See #48.

closed 2019-01-08 by martiniil