PilzDE/pilz_industrial_motion #102: prbt_base is not default reference_frame

martiniil, 2019-02-22:
### Steps to reproduce
1. Modify urdf such that `prbt_base` does not match root-link `world`
2. Execute a ptp pose command without reference frame

### Expected behavior
Robot should move relative to `prbt_base` frame

### Observed behavior
Robot moves relative to `world` frame

martiniil, 2019-02-22:
Is it intended that we require `prbt_base` and `world` to be identical? I am not sure if I fully understand the meaning of these frames. At least it should be possible to position the robot by simply altering the transform from `world` to `prbt_base_link`?
[image]

jschleicher, 2019-02-25:
Please see [ROS-I wiki: Create URDF for an Industrial Robot](http://wiki.ros.org/Industrial/Tutorials/Create%20a%20URDF%20for%20an%20Industrial%20Robot#Additional.2BAC8-Standard_Frames) for a description of standard frames.

Yes, it is possible to move the robot around / move prbt_base to not match the world coordinate system. If you want to command positions in "world" or "base" coordinates in the API is a question of conventions. I'm not aware of a standard there.

[MoveIt commander](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#planning-to-a-pose-goal) refers to a group for every motion plan request; I think that sounds like a good default. Maybe you should choose the default reference frame based on the default planning group as well?

gavanderhoorn, 2019-02-25:
> Please see [ROS-I wiki: Create URDF for an Industrial Robot](http://wiki.ros.org/Industrial/Tutorials/Create%20a%20URDF%20for%20an%20Industrial%20Robot#Additional.2BAC8-Standard_Frames) for a description of standard frames

Slightly off-topic: a more up-to-date (but still WIP) reference: [Coordinate Frames for Serial Industrial Manipulators](https://github.com/gavanderhoorn/rep/blob/coord_frames_ind_robots/rep-xxxx.rst) ([rendered](http://gavanderhoorn.github.io/rep/rep-xxxx.html)).

martiniil, 2019-02-26:
Yes, that sounds very reasonable. Thanks for the input.
Actually, our Readme states that `prbt_base` is used as default.

agutenkunst, 2019-02-27:
Reminder: Increase ROBOT_API if behaviour change on existing programs can be expected.

martiniil, 2019-04-04:
WIP: 7d66d110025c0a768beb7727154fdc70ad957123

hslusarek, 2019-06-06:
Wait for backwards compatibility feature.

commit 34320075 (2019-04-08): Fix transformation of poses in robot api. Fixes #102
commit d50bf76d (2019-04-12): Fix transformation of poses in robot api. Fixes #102
commit 32cf289a (2019-04-12): Fix transformation of poses in robot api. Fixes #102
closed 2023-01-27 by martiniil