PilzDE/pilz_industrial_motion #98: Robot doesn't want to drive to existing cartesian goals

JonathanGruner, 2019-02-22:
If I teached the robot in RVIz and then read the coordinates from tf (see in picture the Terminal in the upper right corner), i save them in Python script. After executing the script tobot says, he could not reach the cartesian position (with joint values it works), see Terminal at the bottom right and at the top left).

I recognized the issue about unnormalized quartenions, and so i did use the exact values as you can see.

[image]

martiniil, 2019-02-22:
@JonathanGruner 
I did execute the following without getting an error:
> r.move(Ptp(goal=Pose(position=Point(-0.478, -0.081, 0.402), orientation=Quaternion(0.044, 0.044, 0.706, 0.706))))

Can you give more details about your settings?
- How did you launch the robot?
- Did you modify the robot description?
- How does your python script look like?

JonathanGruner, 2019-02-22:
@martiniil 
I have given you access at gitlab "ros_workshop" there i described which files i did run.

In general: I did a xacro model of the environment, and inserted an gripper at the robot.
Therefore I copied prbt_moveit_config to my package, and disabled some colissions.

JonathanGruner, 2019-02-22:
The problem was, that I thought the absolute coordinates are relative to the prbt_base_link, but its to the world. And if I move the prbt_base_link, its not working anymore.

Could you maybe append this information in the Pilz robot programming API documentation?
Thanks for fast and efficient support!

closed 2019-02-22 by JonathanGruner