PilzDE/pilz_industrial_motion #110: melodic-ros-base installation: Could not load library...libeigen_conversions.so

rbaur451, 2019-03-12:
### Commit
ros-melodic-pilz-trajectory-generation       0.4.1-1bionic.20190302.180259      armhf

### Steps to reproduce
How can the issue/bug be reproduced?
1. sudo apt install ros-melodic-ros-base 
2. roslaunch prbt_moveit_config moveit_planning_execution.launch pipeline:=pilz_command_planner

### Observed behavior
Error string: Could not load library (Poco exception = libeigen_conversions.so: cannot open shared object file: No such file or directory

### Solution
sudo apt install libtf-conversions-dev

### Reference to test
If there is no test that can detect the issue/bug, please create one and
reference it here.

jschleicher, 2019-03-13:
The ros package `ros-melodic-eigen-conversions` seems to contain that .so as well. Please try to remove `libtf-conversions-dev` and install `ros-melodic-eigen-conversions`.

```
dpkg-query -L ros-melodic-eigen-conversions
/.
/opt
/opt/ros
/opt/ros/melodic
/opt/ros/melodic/include
/opt/ros/melodic/include/eigen_conversions
/opt/ros/melodic/include/eigen_conversions/eigen_kdl.h
/opt/ros/melodic/include/eigen_conversions/eigen_msg.h
/opt/ros/melodic/lib
/opt/ros/melodic/lib/libeigen_conversions.so
/opt/ros/melodic/lib/pkgconfig
/opt/ros/melodic/lib/pkgconfig/eigen_conversions.pc
...
```

The fix would then be to make `eigen_conversions` a run_depend in pilz_trajectory_generation.

rbaur451, 2019-03-13:
Removed the **libtf-conversions-dev** package and installed **ros-melodic-eigen-conversions**.
The solution works fine.

commit 95e73baf (2019-03-18): update dependencies of trajectory_generation

* convert package.xml to format 2 (See http://docs.ros.org/jade/api/catkin/html/howto/format2/migrating_from_format_1.html)
* make eigen_conversions a <depend> to install it at runtime (Closes #110)
* drop prbt_ikfast_manipulator_plugin; trajectory_generation runtime packages
  are independent of the robot used and the corresponding moveit_config is
  already listed as test_depend.
* degrade pilz_testutils to <test_depend>, since in format 2 this also
  fulfils the dependency to build the test (Reverts 421c8af)
* moveit_commander isn't used in traj [... cut]
PR #112 (2019-03-18): update dependencies of trajectory_generation
* convert package.xml to format 2 (See http://docs.ros.org/jade/api/catkin/html/howto/format2/migrating_from_format_1.html)
* make eigen_conversions a `<depend>` to install it at runtime (Closes #110)
* drop prbt_ikfast_manipulator_plugin; trajectory_generation runtime packages
  are independent of the robot used and the corresponding moveit_config is
  already listed as test_depend.
* degrade pilz_testutils to `<test_depend>`, since in format 2 this also
  fulfils the dependency to build the test (Reverts 421c8af)
* moveit_commander isn't used in t [... cut]
commit 93c706b2 (2019-03-18): update dependencies of trajectory_generation

* convert package.xml to format 2 (See http://docs.ros.org/jade/api/catkin/html/howto/format2/migrating_from_format_1.html)
* make eigen_conversions a <depend> to install it at runtime (Closes #110)
* drop prbt_ikfast_manipulator_plugin; trajectory_generation runtime packages
  are independent of the robot used and the corresponding moveit_config is
  already listed as test_depend.
* degrade pilz_testutils to <test_depend>, since in format 2 this also
  fulfils the dependency to build the test (Reverts 421c8af)
* moveit_commander isn't used in traj [... cut]
closed 2019-03-18 by martiniil