PilzDE/pilz_industrial_motion #124: pilz_robot_programming has a lot of unused code

rfeistenauer, 2019-03-20:
### Commit
commit 685c781874d1771b67058b040983813d9185a719

### Steps to reproduce
vulture .../pilz_robot_programming

### Expected behavior
Not much unused stuff

### Observed behavior
quite the opposite. And it doesn't even find all. e.g. _MAX_ACC_SCALE

agutenkunst, 2019-05-16:
Not sure but our coverage analysis showed no problems.

Could you run our coverage tools?

jschleicher, 2019-05-16:
@agutenkunst obviously python-coverage treats object definition as covered and does not require instantiation.
@rfeistenauer for me vulture detected just one exception:
```
vulture --exclude=doc/,test/ .
src/pilz_robot_programming/commands.py:64: Unused attribute 'is_diff'
src/pilz_robot_programming/commands.py:83: Unused attribute 'planning_options'
...
```
Those message fields are sent using `sendGoal` and thus used by the ActionClient. I'd not check for unused variables in tests and thus suggest to close this with #162 (not done with the commit message to let you double-check if you get additional errors).

jschleicher, 2019-05-17:
@rfeistenauer Did you start this issue because of pycharm displaying PEP-8 warnings? Then #164 would help?!

rfeistenauer, 2019-05-20:
No, I just did a vulture check for another PR and it had a lot of warnings, but as you already mentioned, most of them are just ROS messages, that vulture doesn't recognize. #162 should fix this issue

commit cbbba1af (2019-05-23): Drop unused variables in python api

* drop unused exception
* set ptp default velocity as reduced
  correct a typo. PTP shouldn't run with maximum velocity by default.
  Update velocity to 1.0 for tests.
* Drop the unused global variables.
* removed unused imports

Fixes #124
commit 6bdfd3fd (2019-05-23): Drop unused variables in python api

* drop unused exception
* set ptp default velocity as reduced
  correct a typo. PTP shouldn't run with maximum velocity by default.
  Update velocity to 1.0 for tests.
* Drop the unused global variables.
* removed unused imports

Fixes #124
commit 9543fc77 (2019-05-24): Drop unused variables in python api

* drop unused exception
* set ptp default velocity as reduced
  correct a typo. PTP shouldn't run with maximum velocity by default.
  Update velocity to 1.0 for tests.
* Drop the unused global variables.
* removed unused imports

Fixes #124
commit 1422a959 (2019-07-19): Drop unused variables in python api

* drop unused exception
* Drop the unused global variables.
* removed unused imports

Fixes #124
closed 2019-10-11 by ct2034