PilzDE/pilz_industrial_motion #126: LIN Movement: Start point deviates from current robot state

tllatzal, 2019-03-21:
### Commit
0.3.6

### Description
Hello everyone,

I want to use the pilz_industrial_motion package to move a kuka kr16 r1610. I'm using the pilz_robot_programming Python API and the kuka_experimental robot driver. PTP movement works fine but I'm having problems with LIN movement. Things i tried:
1. Absolute linear movement to current postition by inserting "r.get_current_pose" as target pose
2. Absolute linear movement to specific target point within reach
3. Relative linear movement to point within reach
I am always getting the same error that the start point of the trajectory deviates too far from the current robot pose. So I checked the trajectory by looking at the "/move_group/display_planned_path" topic and it seems that the start point indeed is not matching/isn't close to the current robot pose "/joint_states".

### Expected behavior
Linear movement of the robot

### Observed behavior
Error: Invalid Trajectory: Start point deviates from current robot state more than 0.01 joint 'joint_a5": expected: 1.56069, current: 1.5708

### Logfiles
[image]
[image]

martiniil, 2019-03-22:
@tllatzal Thanks for reporting this issue. Can you post more of the log on the lower left side? It only shows output from the second LIN command. It would be good to see also the output from the planning of the first LIN.

tllatzal, 2019-03-22:
@martiniil The error message is the same for all LIN movements. The only difference were the values in the "True cartesian goal". The Python program always throws this error on the first LIN command. No matter if its relative, absolute or absolute to the current position. I also tried omitting the "target_link" parameter without success.

martiniil, 2019-03-22:
Ok. And can you exclude that there is already high deviation in the `/joint_states` themselves?
Then this could be related to the IK solver. I don't know which is used for the kuka in your case. But when the start state deviates drastically from `/joint_states`, I would assume a poor IK solution.

tllatzal, 2019-03-22:
Thank you very much for pointing me in the right direction. The problem was a faulty .urdf file. After recreating it using the .xacro everything seems to work. The robot now moves linearly to the desired cartesian position with absolute accuracy (according to the robot controller). 
I still have a problem with the tool orientation when using the relative parameter and LIN movements. I will look at this after the weekend and maybe open a new issue on that.
Thanks again and have a nice weekend

closed 2019-03-22 by tllatzal