PilzDE/pilz_industrial_motion #127: Blending issue: Trajectory message contains waypoints that are not strictly increasing in time

tllatzal, 2019-03-26:
### Expected behavior
Blended linear motion between two points.

### Observed behavior
Moveit Error:
[ERROR] [1553595062.866247519]: Trajectory message contains waypoints that are not strictly increasing in time.

## Description
Hello everyone,

when I try to move the robot (Kuka KR16 R1610) using blended linear motion I get the error above by MoveIt. So I checked the trajectory being send to the robot driver (/joint_trajectory_action/goal). The blending trajectory begins at point 203 and ends after 42 points at point 245 (if I understand the Moveit-Log correctly). There are a few things i noticed:
1. The "time_from_start" parameter increases constantly at the transition from the first trajectory to the blending trajectory.
2. The "time_from_start" parameter is exactly the same for two points at the transition from the blending trajectory to the second trajectory which probably explains the MoveIt error message.
3. Those two points have zero velocity and zero acceleration (which is not the point of blending, I guess)

I added a few excerpts from the terminal logs and the python program being used.
If you need any more information I would be happy to give it to you. 

When setting the blending radius to zero there is no error and all points can be reached. I tried to use different positions and different radii without success.

Thanks for your help in advance

### Additional Information
**Moveit Log:**
[ INFO] [1553595062.808083929]: Starting generation of LIN Trajectory!
[ INFO] [1553595062.829076769]: LIN Trajectory with 246 Points generated. Took 20.9675 ms.
[ INFO] [1553595062.830782339]: Starting generation of LIN Trajectory!
[ INFO] [1553595062.848031139]: LIN Trajectory with 246 Points generated. Took 17.2255 ms.
[ INFO] [1553595062.849797059]: Starting generation of LIN Trajectory!
[ INFO] [1553595062.856551729]: LIN Trajectory with 95 Points generated. Took 6.72828 ms.
[ INFO] [1553595062.857332949]: Start trajectory blending using transition window.
[ WARN] [1553595062.857349709]: This signature of isRobotStateEqual is deprecated. Please use the new one in the future.
[ INFO] [1553595062.857379629]: Search for start and end point of blending trajectory.
[ INFO] [1553595062.857513609]: Intersection point of first trajectory found, index: 203
[ INFO] [1553595062.857635549]: Intersection point of second trajectory found, index: 42
[ERROR] [1553595062.866247519]: Trajectory message contains waypoints that are not strictly increasing in time.

**Python Log**
```
[INFO] [1553595062.754794]: Move: Sequence
[ERROR] [1553595063.078297]: Failure during execution of: Sequence:
[...coordinates and orientation...]
Traceback (most recent call last):
  File "/home/hutab/hutab_ws/src/pilz_industrial_motion/pilz_robot_programming/examples/kr16_test.py", line 84, in <module>
    start_program()
  File "/home/hutab/hutab_ws/src/pilz_industrial_motion/pilz_robot_programming/examples/kr16_test.py", line 77, in start_program
    r.move(sequence)
  File "/home/hutab/hutab_ws/src/pilz_industrial_motion/pilz_robot_programming/src/pilz_robot_programming/robot.py", line 228, in move
    self._move_execution_loop(cmd)
  File "/home/hutab/hutab_ws/src/pilz_industrial_motion/pilz_robot_programming/src/pilz_robot_programming/robot.py", line 317, in _move_execution_loop
    raise RobotMoveFailed("Failure during execution of: " + str(cmd))
pilz_robot_programming.exceptions.RobotMoveFailed: Failure during execution of: Sequence:
[...coordinates and orientation...]
```
**topic: /joint_trajectory_action/goal**
**Points 202, 203, 204**
```
        positions: [7.593925488436071e-14, -1.5507024864834043, 1.814085449169146, 3.879867859561321e-07, 1.3074128021973346, -2.883150902542115e-07]
        velocities: [-3.1086244689504383e-13, 0.02759764999429315, 0.11736654236833566, -1.7784491303317022e-06, -0.1449615937693105, 1.2724071932407696e-06]
        accelerations: [0.0, 0.0013100016249545376, -0.09087688061404364, 9.614615524245476e-07, 0.08956528286585552, -2.427579160328893e-07]
        effort: []
        time_from_start: 
          secs: 2
          nsecs:  10000000
      - 
        positions: [7.283063041541027e-14, -1.5504307905977273, 1.8152316667749433, 3.705782708578284e-07, 1.3059949141239195, -2.758173764180495e-07]
        velocities: [-3.1086244689504383e-13, 0.027169588567699243, 0.11462176057972506, -1.740851509830374e-06, -0.141788807341503, 1.249771383616197e-06]
        accelerations: [0.0, -0.042806142659390645, -0.2744781788610595, 3.7597620501328217e-06, 0.31727864278074946, -2.263580962457265e-06]
        effort: []
        time_from_start: 
          secs: 2
          nsecs:  20000000
      - 
        positions: [6.927791673660977e-14, -1.550163711633907, 1.8163496393544953, 3.535604436637131e-07, 1.3046098874103318, -2.6355948767297716e-07]
        velocities: [-3.552713678800501e-13, 0.02670789638201665, 0.11179725795520401, -1.7017827194115264e-06, -0.1385026713587756, 1.2257888745072356e-06]
        accelerations: [-4.440892098500626e-12, -0.046169218568259396, -0.28245026245210525, 3.906879041884768e-06, 0.3286135982727423, -2.398250910896134e-06]
        effort: []
        time_from_start: 
          secs: 2
          nsecs:  30000000
      - 
```
**Points 245, 246, 247, 248**
```
        positions: [-3.1086244689504383e-15, -1.5442944096275864, 1.8390054060235241, 1.0266425594839958e-10, 1.2760853302516018, -7.894210513900487e-11]
        velocities: [0.0, 0.0008638215248879533, 0.0030975518777731637, -4.9501264929535425e-08, -0.003961302350807827, 3.805990649010882e-08]
        accelerations: [0.0, -0.07148535256185085, -0.25649769632707375, 4.098157686102605e-06, 0.32797716614130223, -3.1501168515242405e-06]
        effort: []
        time_from_start: 
          secs: 2
          nsecs: 440000000
      - 
        positions: [-7.105427357601002e-15, -1.5442926179448366, 1.839011829987876, 1.2798368348577409e-16, 1.2760771147518575, -3.712353924600952e-15]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 2
          nsecs: 447078000
      - 
        positions: [-1.4210854715202004e-14, -1.544292617945806, 1.8390118299888643, 1.279836835242051e-16, 1.276077114746942, -7.509996963242089e-15]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 2
          nsecs: 447078000
      - 
        positions: [-1.3219741493486481e-05, -1.5442926178506118, 1.8390118298907367, 1.95630454774057e-16, 1.2760771147498753, -1.3219741486805273e-05]
        velocities: [-0.0013219741479275626, 9.519429688964465e-09, -9.812750612070431e-09, 6.7646771249851905e-15, 2.9332092310596636e-10, -0.0013219741479295276]
        accelerations: [-0.13219741479275626, 9.519429688964465e-07, -9.812750612070431e-07, 6.76467712498519e-13, 2.9332092310596636e-08, -0.13219741479295277]
        effort: []
        time_from_start: 
          secs: 2
          nsecs: 457078000
      - 
```

**Python Program**
```
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1))
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.9, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1))
    r.move(sequence)
```

tllatzal, 2019-03-26:
I just found out that it does work if the first Lin command in the sequence is also blended.
So this works:
```    
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.9, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1))
```
This doesn't:
```    
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1))
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.9, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1))
```
It seems generally blending does not work if it is being started within a sequence where the previous point was not blended. Because this also does not work for me:
```
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1))
    sequence.append(Lin(goal=Pose(position=Point(0.851, 0, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1), blend_radius=0.02)
    sequence.append(Lin(goal=Pose(position=Point(0.9, 0.2, 1.1), orientation=from_euler(math.radians(180), math.radians(0), math.radians(180))), target_link="tool0", vel_scale=0.1, acc_scale=0.1))
```

I don't know if this is a bug, wanted behaviour or something which is not implemented yet.

martiniil, 2019-04-18:
Blending should work now for all cases. (Or at least with next release)

closed 2019-04-18 by martiniil