PilzDE/pilz_industrial_motion #128: pilz_command_planner: Rviz Plugin - Movement not possible

hslusarek, 2019-03-26:
### Commit
- [ ] @hslusarek TODO Add commit.

### Steps to reproduce
Launch real robot via:  
```roslaunch prbt_moveit_config moveit_planning_execution.launch sim:=false pipeline:=pilz_command_planner```  
In Rviz perform the following steps:
1. Select Ptp command.
2. Choose position to move to.
3. Push "Plan and Execute" button and wait for robot to complete motion.
4. If "Plan and Execute" button is active, after motion is finished, go back to step 2.

### Expected behavior
After motion is finished, it should be possible to plan and execute a new motion.

### Observed behavior
For some reason sometimes the "Plan and Execute" button remains deactivated. Maybe the Pluging still waits for the motion to finish and has not received the finish signal. On the same day but maybe not in this context I also had problems with a statement like ```PATH_TOLERANCE_VIOLATED```. Maybe it is related? I normally resolved my problems by restarting ROS. The I was often able to again perform one or two Ptp movements until the problem reappeared.

hslusarek, 2019-03-28:
- [ ] Reproduce bug in lab.

hslusarek, 2019-04-26:
- [ ] TODO: Determine Firmeware version of robot on which bug occured

hslusarek, 2019-05-17:
Bug could **not** be reproduced with:
- kinetic
- pilz_industrial_motion - Version: 0.3.6-0
- pilz_robots - Version: 0.4.7-0
- prbt_grippers - Version: 0.0.2-0
- Firmeware: 101 Build:11345 Date:2018-12-17 07:33:06\x15\x02\0

martiniil, 2019-05-22:
Close?

hslusarek, 2019-06-06:
Yes. Close because not reproducible.

closed 2019-06-06 by hslusarek