PilzDE/pilz_industrial_motion #154: Guessing parameters

JonathanGruner, 2019-04-30:
I have a problem with programming a application effective, because of guessing parameters if the program is not able to execute the move commands. I would request a feature to make programming more usable for industries if the parameters are not important for me but for the program, than the program should use a possible parameter on his own.

i.e.: LIN movements--> velocity. If i don't give any velocity parameter than i dont care about velocity, the robot should drive as fast as he can.

It would help too if the program could recommend me a working parameter, even this is way faster as to step yourself to the right parameter, because in most cases I really don't care about them.

This includes:
- LIN-->velocity, acceleration
- Sequence --> Blend radius, acceleration

[image]
[image]

hslusarek, 2019-04-30:
@dbakovic Please evaluate this as possible new feature.

dbakovic, 2019-05-16:
@JonathanGruner Thank you for your suggestion. We have it already on our todo list. Due to tasks of higher priority it will take a while before we start with the implementation.

closed 2023-01-27 by martiniil