PilzDE/pilz_industrial_motion #185: pilz_robot_programming tests fail on travis

martiniil, 2019-07-09:
### Commit
11d3618371d7df8e4bc186c80cbf8bd773cc08c8

### Expected behavior
Tests are executed without error

### Observed behavior
Execution fails due to import error

### Logfiles

>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 41, in <module>
> 
> 
>   from moveit_ros_planning_interface import _moveit_move_group_interface
> 
> 
> ImportError: libeigenpy.so: cannot open shared object file: No such file or directory

### Reference to test
pilz_robot_programming/test/integrationtests/tst_api_execution_stop.py

martiniil, 2019-07-09:
triggered by latest kinetic release of moveit. Related issue:
https://github.com/ros-planning/moveit/issues/1536

jschleicher, 2019-07-09:
Not a problem in pilz_robot_programming. Reported upstream. Closing here.

closed 2019-07-09 by jschleicher