PilzDE/pilz_industrial_motion #208: fatal error: eigen3/Eigen/Eigen:

dabhijeet, 2019-09-26:
Hello i downloaded the pilz_industrial_motion-kinetic-devel. When i am doing catkin_make i get the following error --

fatal error: eigen3/Eigen/Eigen: No such file or directory
compilation terminated.

I also tried doing the same in another laptop and this problem didn't occur there. I don't know why it is occurring in my Laptop.

martiniil, 2019-09-26:
@dabhijeet Did you install all dependencies? This is usually done with rosdep.

dabhijeet, 2019-09-27:
hello,
I tried these commads but nothing happened
sudo apt-get update
sudo apt-get upgrade
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y
then i opened Synaptic Package Manager and searched for eigen and selected everything for installation and the **error is gone**......
Now there is no problem during catkin_make

closed 2019-09-27 by dabhijeet