PilzDE/pilz_industrial_motion #210: Integration with other robot

dabhijeet, 2019-09-27:
Hello,

I was trying to integrate pilz_industrial_motion with our package. For that we changed the below parameters according to our robot

_DEFAULT_PLANNING_GROUP = "arm"
_DEFAULT_TARGET_LINK = "tcp"
_DEFAULT_GRIPPER_PLANNING_GROUP = "gripper"
_DEFAULT_BASE_LINK = "Link_1"

Also we commented the exceptions.py so that there won't be error related to BrakeTestErrorCodes. 

But when we run 
rosrun pilz_robot_programming demo_program.py
it is waiting for the action server (**Waiting for connection to action server sequence_move_group...**) 

I am attaching the rqt_graph result. Please let me know what mistake i did in the process..

Thank you in advance

[image]

martiniil, 2019-09-30:
When you start the move_group you have to specify our planner `pilz::CommandPlanner` and the capabilities `pilz_trajectory_generation/MoveGroupSequenceAction` and `pilz_trajectory_generation/MoveGroupSequenceService`.

See https://github.com/PilzDE/pilz_industrial_motion/blob/melodic-devel/pilz_trajectory_generation/README.md#using-the-command-planner

The `move_group.launch` for our robot shows how this is done:
https://github.com/PilzDE/pilz_robots/blob/melodic-devel/prbt_moveit_config/launch/move_group.launch

dabhijeet, 2019-09-30:
Thanks....
Now it is working fine..

closed 2019-09-30 by dabhijeet

logicraju, 2022-02-21:
> Thanks.... Now it is working fine..

How did you solve it ? Regards