PilzDE/pilz_industrial_motion #211: Issue in sequence command

dabhijeet, 2019-10-01:
Hello,

I was testing the ptp, lin and sequence commands in kinetic with ompl planning. ptp and lin is working fine in my case but sequence command is not working..
When i run 

    sequence = Sequence()
    sequence.append(Ptp(goal=[1, 0, 0, 1.57079632, 0, 0], vel_scale=0.4))
    sequence.append(Ptp(goal=[0, 1, 0, 1.57079632, 0, 0], vel_scale=0.4))
    r.move(sequence)

commands

the robot stops at the last point before sequence moves and i get 

[INFO] [1569926254.891125, 167.917000]: Move: Sequence

in terminal.

Also in the moveit terminal i get - 

[ INFO] [1569926255.330609972, 168.145000000]: Solution found in 0.164460 seconds
[ INFO] [1569926255.518818557, 168.260000000]: SimpleSetup: Path simplification took 0.184086 seconds and changed from 4 to 2 states
/opt/ros/kinetic/lib/moveit_ros_move_group/move_group: symbol lookup error: /home/ros/brabo_catkin/devel/lib//libsequence_capability.so: undefined symbol: _ZN4pilz17isRobotStateEqualERKN6moveit4core10RobotStateES4_RKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEd
[move_group-3] process has died [pid 5360, exit code 127, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/ros/.ros/log/e6b2499e-e436-11e9-9046-5ce0c5178815/move_group-3.log].
log file: /home/ros/.ros/log/e6b2499e-e436-11e9-9046-5ce0c5178815/move_group-3*.log

Any idea what can be the reason for it

jschleicher, 2019-10-01:
The sequence needs a sequence capability that is included in `pilz_trajectory_generation` and thus only works with `pilz_command_planner` (not OMPL). See the tutorials [Move Robot with pilz_command_planner](https://wiki.ros.org/pilz_robots/Tutorials/MoveRobotWithPilzCommand_planner) and following for installation and usage of the planner.

dabhijeet, 2019-10-03:
yeah...sequence is not working with OMPL. With pilz_command_planner it is working fine...

closed 2019-10-03 by dabhijeet