PilzDE/pilz_industrial_motion #214: pause, resume , stop does not work with sequence command.

nikitapurandare, 2019-10-04:
Hello,
While running Lin or Ptp commands, we observed that pause ,resume, play i.e 
rosservice call /pause_movement 
rosservice call /resume_movement 
rosservice call /stop_movement 
worked well with the robot. But when we used it with sequence commands we got the following output in the terminal :

[ INFO] [1570172166.302663050]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1570172166.676782963, 2625.534000000]: IK Using joint Link_2 -3.14159 3.14159
[ INFO] [1570172166.676850147, 2625.534000000]: IK Using joint Link_3 -1.5708 1.5708
[ INFO] [1570172166.676885830, 2625.534000000]: IK Using joint Link_4 -2.35619 1.0472
[ INFO] [1570172166.676915954, 2625.534000000]: IK Using joint Link_5 -1.91986 1.91986
[ INFO] [1570172166.676949468, 2625.534000000]: IK Using joint Link_6 -6.28319 6.28319
[ INFO] [1570172166.676984683, 2625.534000000]: IK Using joint tcp -1.4 1.4
[ INFO] [1570172166.677644456, 2625.534000000]: Looking in common namespaces for param name: arm/position_only_ik
[ INFO] [1570172166.685238520, 2625.540000000]: Looking in common namespaces for param name: arm/solve_type
[ INFO] [1570172166.694762509, 2625.543000000]: Using solve type Speed
[INFO] [1570172166.735244, 2625.292000]: RobotCommander created.
[ INFO] [1570172167.852433552, 2626.376000000]: Ready to take commands for planning group arm.
[INFO] [1570172171.339783, 2628.818000]: Pause called.
[INFO] [1570172171.342403, 2628.819000]: Switching state from NO_REQUEST to PAUSE_REQUESTED
[ERROR] [1570172171.692033, 2629.049000]: Pause not implemented for sequence yet

and this is output of moveit terminal : 

[ INFO] [1570176546.122540044, 6694.678000000]: Combined planning and execution request received for MoveGroupSequenceAction.
[ INFO] [1570176546.122914631, 6694.678000000]: Planning attempt 1 of at most 1
[ INFO] [1570176546.123964193, 6694.679000000]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1570176546.124268685, 6694.679000000]: Generating PTP trajectory...
[ INFO] [1570176546.134131039, 6694.685000000]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1570176546.134319843, 6694.685000000]: Generating PTP trajectory...
[ INFO] [1570176546.142995843, 6694.688000000]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1570176546.143256036, 6694.688000000]: Generating PTP trajectory...
[ INFO] [1570176546.152426565, 6694.691000000]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1570176546.152567652, 6694.691000000]: Generating PTP trajectory...
[ INFO] [1570176552.038997040, 6699.282000000]: Stopping execution due to preempt request
[ INFO] [1570176552.039097855, 6699.282000000]: Cancelling execution for brabotr105/joint_trajectory_controller
[ INFO] [1570176552.039187089, 6699.282000000]: Stopped trajectory execution.
[ INFO] [1570176552.041528868, 6699.282000000]: Completed trajectory execution with status PREEMPTED ...

Can we pause , resume or stop while executing sequence commands or is there any other solution for this ?

hslusarek, 2019-10-04:
> Can we pause , resume or stop while executing sequence commands or is there any other solution for this ?

Hello. 
To answer your question directly: Right now the pause is not supported for a sequence command. You also see it on your terminal: `Pause not implemented for sequence yet`.
*Possible other solution:*
If you want to be able to pause your application at an arbitrary point of execution, I suggest you use only single commands in your application. If you only have certain movements which you want to be able to pause, I suggest you make these movements to single commands (in other words, do not put them in a sequence). The movements which do not need to be paused can remain in sequences.

Does this answer help you?

nikitapurandare, 2019-10-04:
Yes ...
Thanks a lot .We just wanted to know if you are planning to include this feature and relative command in sequence .

closed 2023-01-27 by martiniil