PilzDE/pilz_industrial_motion #218: Configuring planner

dabhijeet, 2019-10-11:
How can we configure pilz_command_planner?
(Like OMPL can be configured by changing ompl_planning.yaml)

agutenkunst, 2019-10-15:
Hi @dabhijeet ,
you need to load the capabilties 
`pilz_trajectory_generation/MoveGroupSequenceAction pilz_trajectory_generation/MoveGroupSequenceService` and pass it to the move_group as arg.

Furthermore you need 
https://github.com/PilzDE/pilz_robots/blob/melodic-devel/prbt_moveit_config/launch/pilz_command_planner_planning_pipeline.launch.xml placed in your moveit_config/launch folder.

If you send me a link to your robot package maybe I can help integrating it.

dabhijeet, 2019-10-17:
Hi,
We have already loaded the above capabilities. 
When we were using OMPL we got to know that each parameter of ompl_planning.yaml needed to be fine tuned in order to get rid of vibration and have smooth motion. (i.e. longest_valid_segment_fraction, maximum_waypoint_distance, max_sampling_attempts etc.)
But in pilz command planner there is cartesian_limits.yaml and start_state_max_bounds_error parameter and by changing these parameters also we are not getting smooth motion in linear. so that is why we wanted to know how to configure pilz command planner.

jschleicher, 2020-04-20:
Probably solved after #213.

closed 2020-04-20 by jschleicher

logicraju, 2022-02-21:
> Hi @dabhijeet , you need to load the capabilties `pilz_trajectory_generation/MoveGroupSequenceAction pilz_trajectory_generation/MoveGroupSequenceService` and pass it to the move_group as arg. [...]

Hi, i'm  trying to integrate it in UR10. Could you help me please ? Regards