PilzDE/pilz_industrial_motion #235: Trigger another robot movement process via sensor signals

CloudYin, 2020-01-13:
**Is your feature request related to a problem? Please describe.**
In the laser scanner callback function, I want to implement a code that the robot move in a new path after an area of scanner is intruded. If I use some move command in scanner callback function. A RobotMoveAlreadyRunningError error will appear:

[ERROR] [1578287684.790064]: bad callback: <function laser_scan_callback at 0x7f011a5ca850>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/pilz/catkin_ws/src/relay_pick_and_place/scripts/robot_test.py", line 19, in laser_scan_callback
    r.move(Ptp(goal=home_pos, vel_scale=__MOVE_VELOCITY__))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/pilz_robot_programming/robot.py", line 236, in move
    raise RobotMoveAlreadyRunningError("Parallel calls to move are note allowed.")
RobotMoveAlreadyRunningError: Parallel calls to move are note allowed.

**Describe the solution you'd like**
A simple process should like this:
1. The robot moves in default path A after start-up.
2. If an area is intruded (or another sensor is set), process A stop, robot moves in path B. 
3. When the area is clean, process B stop, robot back to path A.  

**Describe alternatives you've considered**
None

agutenkunst, 2020-01-13:
Hi!
We (@agutenkunst  and @hslusarek ) had a look at it and can provide you with a "hacky" solution for now
```
#!/usr/bin/env python
# Copyright (c) 2020 Pilz GmbH & Co. KG
#
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU Lesser General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU Lesser General Public License for more details.
#
# You should have received a copy of the GNU Lesser General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

from geometry_msgs.msg import Point
from pilz_robot_programming.robot import *
from pilz_robot_programming.commands import *
import math
import rospy
from std_srvs.srv import Trigger
import threading

__REQUIRED_API_VERSION__ = "1"

r = None

class HackRobot(Robot):
    def parallelMove(self, cmd):
	self._sequence_client.send_goal(cmd._get_sequence_request(self))
    	self._sequence_client.wait_for_result()

def _pause_callback(request):
    # Pause needs to be called from another thread
    pause_thread = threading.Thread(target=r.pause)
    pause_thread.start()

    # Make sure robot has stopped
    rospy.sleep(.3)

    # Your motion command
    r.parallelMove(Ptp(goal=[0, 0, 0, 0, 0, 0], vel_scale=0.4))
    r.parallelMove(Ptp(goal=[0.1, 0, 0, 0, 0, 0], vel_scale=0.4))

    pause_thread.join()
    return [True, "success"]

def _resume_callback(request):
    rospy.logerr("_resume_callback()")
    r.resume()
    return [True, "success"]

# The path A
def start_program(r):
    print("Executing " + __file__)

    # Simple ptp movement
    while not rospy.is_shutdown():
        r.move(Ptp(goal=[0, 0.5, 0.5, 0, 0, 0], vel_scale=0.4))
        r.move(Ptp(goal=[0, -0.5, 0.5, 0, 0, 0], vel_scale=0.4))

if __name__ == "__main__":
    # Init a ros node
    rospy.init_node('robot_program_node')

    r = HackRobot(__REQUIRED_API_VERSION__)

    pause_service = rospy.Service("move_path_B", Trigger, _pause_callback)
    resume_service = rospy.Service("continue_move_path_A", Trigger, _resume_callback)

    start_program(r)
```
please note that this is really a hack skipping all checks that exist for various reasons.

We will try to find a more comprehensive solution to allow such parallel and reactive behavior via python API!

Please let us know if this solution works for you for now.

CloudYin, 2020-01-19:
Dear both (@agutenkunst and @hslusarek) ,

here is my test code:
[https://github.com/CloudYin/psen_scan_application/blob/master/scripts/psen_scan_application.py](url)

when debugging the following error appeared:

pilz@pilz:~/catkin_ws$ rosrun psen_scan_application psen_scan_application.py 
  File "/home/pilz/catkin_ws/src/psen_scan_application/scripts/psen_scan_application.py", line 40
    self._sequence_client.wait_for_result()
                                          ^
IndentationError: unindent does not match any outer indentation level

No idea how to fix this issue. I use tabs instead of spaces in the code.

CloudYin, 2020-01-19:
Dear both  (@agutenkunst and @hslusarek ), here are my test scripts:
[https://github.com/CloudYin/psen_scan_application/tree/master/scripts](url)

I made a following test:
1. First run the robot_application.py. In a new terminal call the /move_path_B or /continue_move_path_A, both functions passed.
2. Then run the psen_scan.py. Try to call both services in the scanner callback, failed and prompted the following python error:
[ERROR] [1579420263.697772]: bad callback: <function laser_scan_callback at 0x7f3ddb51df50>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/pilz/catkin_ws/src/psen_scan_application/scripts/psen_scan.py", line 14, in laser_scan_callback
    rospy.wait_for_service('move_path_B', Trigger)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 126, in wait_for_service
    timeout_t = time.time() + timeout
TypeError: unsupported operand type(s) for +: 'float' and 'type'

Don't understand what's wrong in the second script.

agutenkunst, 2020-02-04:
hi @CloudYin 
seems the second parameter of `'wait_for_service` is the timeout you want to wait for the service.

See also
http://docs.ros.org/api/rospy/html/rospy.impl.tcpros_service.ServiceProxy-class.html
you provide "Trigger" which is a type. Either provide just one parameter to wait infinitely or put e.g. 0.5 to wait 0.5 seconds for the service.

CloudYin, 2020-02-11:
> hi @CloudYin
> seems the second parameter of `'wait_for_service` is the timeout you want to wait for the service.
>
> See also [...]

Hi Alex @agutenkunst 

Thanks for the correction. To simplify the test I deleted the wait_for_service call and updated the script:
[https://github.com/CloudYin/psen_scan_application/blob/master/scripts/psen_scan.py]
The execution is ok, however it causes a big delay in executing the pause/resume trigger. I think because I put both of them in the laser scan callback function, so that it repeatedly execute the trigger in each callback message. Is there a good way to improve my code?

agutenkunst, 2020-02-17:
You could try react only on "Flips" in the distance. That would require to store the last distance value and compare it to the current value. If the distance "flipped" trigger the service call. Please let me know if this works for you.

CloudYin, 2020-02-21:
Thanks Alex that works. 
Now the sequence move in path B is successful. But if I put the movement in a while loop in path B, such as:

# Your motion command
while not rospy.is_shutdown():
    r.parallelMove(Ptp(goal=[0, 0, 0, 0, 0, 0], vel_scale=0.4))
    r.parallelMove(Ptp(goal=[0.1, 0, 0, 0, 0, 0], vel_scale=0.4))

then it cannot go back to path A anymore.

agutenkunst, 2020-04-29:
@jschleicher Could you help formulate this into a feature request to help @CloudYin ?

closed 2020-07-21 by CloudYin