PilzDE/pilz_industrial_motion #236: Allow joint configurations to be defined in degree

HiddenBug, 2020-01-15:
### Observed behavior
Test with the real system and start-up engineers have shown that it is unintuitive for "normal" robot users to specify a joint configuration using rad instead of degree. Furthermore, it is confusing for "normal" robot users to "guess" the unit in which a certain number is expressed. The code should express the unit which is used.

### Expected behavior
I expect the PythonAPI to provide additional functions/function signatures for the definition of a joint configuration allowing the user to easily define joint configurations using degree instead of rad. Furthermore, the code should explicitly express the unit in which a number is stated.

Currently, joint configurations in commands are defined like this:
```
r.move(Ptp(goal=[0, 0.5, 0.5, 0, 0, 0], vel_scale=0.4))
```
Additionally, to the above stated definition I suggest to provide functions/functions signatures which allow the user to define joint configurations, e.g. like this:
```
r.move(Ptp(goal=deg([0, 28.65, 28.65, 0, 0, 0]), vel_scale=0.4))
# or:
r.move(Ptp(goal=rad([0, 0.5, 0.5, 0, 0, 0]), vel_scale=0.4))
```

gavanderhoorn, 2020-01-15:
As an external observer here: ROS uses radians for all angles (REP-3). Many robot OEMs have different conventions for their (joint) pose formats (degrees, radians, millimetres, metres, euler angles, quaternions, axis-angle, etc).

It's not so unexpected that degrees are not used in my opinion, but it should be clearly documented. This would seem like a normal thing for users to need to get familiar with.

Would that not be sufficient?

jschleicher, 2020-04-20:
That's why we suggest `math.radians` inside [prbt_default_application](https://github.com/PilzDE/pilz_application_templates/blob/497306052d37ac0603673af2f2f35ddc2beaaadd/prbt_default_application/scripts/myApplication.py#L22-L23): 

```
    pick_pose = Pose(position=Point(0, -0.5, 0.25),  # cartesian coordinates
                     orientation=quaternion_from_euler(
                       0,
                       math.radians(180),
                       math.radians(0)))
```

So I agree with @gavanderhoorn that this is mostly a documentation issue. @HiddenBug Would you mind, preparing a PR for the demo program and Readme here, demonstrating the use of `math.radians`?

closed 2023-01-27 by martiniil