PilzDE/pilz_industrial_motion #238: Document how to set default commands private variables, _DEFAULT_TARGET_LINK, _DEFAULT_BASE_LINK etc.

garyedwards, 2020-02-05:
**Is your feature request related to a problem? Please describe.**

I am trying to use the Pilz path planner with an ABB robot.  At the moment I can not find a way to initialise the commands object with different values for `_DEFAULT_PLANNING_GROUP = "manipulator" _DEFAULT_TARGET_LINK = "prbt_tcp" _DEFAULT_GRIPPER_PLANNING_GROUP = "gripper" _DEFAULT_BASE_LINK = "prbt_base"` etc.

This seems quite important when working with different models with different naming standards.  Also if you have multiple manipulators these would likely be namespaced and again you may want to initialise the commands object with different values in the same script for example.

It may be a lack of Python knowledge but I am struggling to find a way to get this to work.  If this can be document in comments of this issue initially and then on the wiki that would help others like me.

**Describe the solution you'd like**

Document how to set commands private variables as described above for robots with different naming conventions or that have been namespaced.

**Describe alternatives you've considered**

Alternatives are setting up an xacro with fixed joints of the expected names and dedicated moveit config to conform to the pilz naming conventions.  This works fine for a single robot but does not solve the namespacing issue described.

**Additional context**

Thanks for a great project I look forward to hearing your thoughts.

garyedwards, 2020-02-05:
These test scripts are a very handy ref:

https://github.com/PilzDE/pilz_industrial_motion/tree/melodic-devel/pilz_trajectory_generation/test/test_robots

garyedwards, 2020-02-05:
OK the `testpoints.py` file linked above answered my questions.  I still think this is an area were the documentation could be improved.  I will close this issue and hopefully it will help others find the above useful directory.

closed 2020-02-05 by garyedwards

agutenkunst, 2020-02-05:
hi @garyedwards thanks for reaching out to us.
Our buildjob for pilz_robot_programming is currently broken and thus the documentation was incomplete. Sorry for this.

But even with a proper documentation I agree that we should have a look into the documentation of these features.