PilzDE/pilz_industrial_motion #241: pilz_robot_programming documentation of private classes is missing

agutenkunst, 2020-02-05:
### Commit
e11efa059f44bb3402770cae2ab56d7b46dccd48
Note that: #239 is needed for proper documentation generation

### Description
If a class has a underscore prefix no documentation is generated for it.
Thus e.g. `_BaseCmd` has no documentation generated.
This can cause things like #238 

Maybe @tfoote or @jobleier have any input on this?

agutenkunst, 2020-02-06:
I am not able to figure out some decent way and therefore would suggest a rename `_BaseCmd` to `BaseCmd` @hslusarek your opinion?

hslusarek, 2020-02-17:
> I am not able to figure out some decent way and therefore would suggest a rename `_BaseCmd` to `BaseCmd` @hslusarek your opinion?

I think your suggestion to rename `_BaseCmd` to `BaseCmd` is ok. Maybe at one point or another an external programmer wants to add to our commands and then this change might also come in handy.

PR #249 (2020-02-27): Rename _BaseCmd -> BaseCmd
Fixes #241 . Documentation is generated for BaseCmd.
closed 2020-02-27 by hslusarek