PilzDE/pilz_industrial_motion #257: Planner does not seem to recognize the cartesian limits

ipa-rmb-mf, 2020-04-01:
Hello,

I am trying to use the PTP planning algorithm via ROS/RVIZ on another MoveIt configuration than the PRBT one. As everything works with the OMPL planner delivered with MoveIt, I assumed that there is no basic mistake within my moveit_config files.
In order to use the Pilz planners, I have made the necessary changes within my configuration, in particular also to add the cartesian limits yaml file and the corresponding lines in planning_context.launch. I can run the configuration, RVIZ loads the scene, I can manipulate the desired goal pose of my robot and the PTP planner can be chosen in the MoveIt side panel, alongside the CIRC and the LIN planner. However, once I try to start the planner via the MoveIt side panel, the planner fails with the error message `acceleration limit not set for group [name of my group]`, likely thrown through line 53 in [trajectory_generator_ptp.cpp](http://docs.ros.org/melodic/api/pilz_trajectory_generation/html/pilz_trajectory_generation/trajectory__generator__ptp_8cpp_source.html). This happens although all necessary cartesian limits can be found in the parameter server through the correct address `/robot_description_planning/cartesian_limits/max_rot_vel, [...]`, as can be seen in the terminal output. 

Now, the question is what could be the reason of this error? My impression is that something in my SRDF file for the group description might be incompatible, as it seems to be somehow related to my move group. However, everything appears to be correct. Is that a known fail scenario and is there a fix?

Best regards
Martin F.

martiniil, 2020-04-02:
The error message indicates that joint limits are missing.

Can you check your joint limits configuration?
For an example see: https://github.com/PilzDE/pilz_robots/blob/melodic-devel/prbt_moveit_config/config/joint_limits.yaml

ipa-rmb-mf, 2020-04-02:
Thank very much, this resolved the issue.
In the joint_limits.yaml, the `has_acceleration_limits` values of each joint were set to false. The planner now operates as expected.

closed 2020-04-02 by ipa-rmb-mf