PilzDE/pilz_industrial_motion #263: Python API segmentation fault

SansoneG, 2020-04-22:
### Commit
59f80849ff

### Steps to reproduce
1. Execute demo_program.py

### Observed behavior
Everything is excecuted as expected, but at the end segmentation fault is reported.

### Logfiles
```
Executing /home/gsansone/catkin_ws/src/pilz_industrial_motion/pilz_robot_programming/examples/demo_program.py
[INFO] [1587549598.452606]: Waiting for connection to action server sequence_move_group...
[INFO] [1587549598.696596]: Waiting for connection to service /prbt/get_speed_override...
[INFO] [1587549598.702237]: Connection to service /prbt/get_speed_override estabilshed
[INFO] [1587549598.705450]: Move: Ptp
[ERROR] [1587549598.729616803]: Virtual joint does not attach to a link on the robot (link 'world' is not known)
[ INFO] [1587549598.731583855]: Loading robot model 'prbt'...
[ INFO] [1587549598.731655205]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1587549598.774270665]: Using link_prefix: 'prbt_'
[INFO] [1587549598.778157]: RobotCommander created.
[ INFO] [1587549599.985323657]: Ready to take commands for planning group manipulator.
[INFO] [1587549602.481543]: Move: Ptp
[INFO] [1587549604.523664]: Move: Ptp
[INFO] [1587549605.809656]: Move: Ptp
[INFO] [1587549607.062826]: Move: Lin
[INFO] [1587549608.918050]: Move: Lin
[INFO] [1587549611.582124]: Move: Lin
[INFO] [1587549614.251475]: Move: Circ
[INFO] [1587549616.921181]: Move: Ptp
[INFO] [1587549622.365104]: Move: Sequence
[INFO] [1587549631.075637]: Move: Ptp
[INFO] [1587549632.337130]: Move: Sequence
[INFO] [1587549637.470976]: Move: Ptp
[INFO] [1587549638.520444]: Move: Ptp
[INFO] [1587549639.361015]: Move: Ptp
[ERROR] [1587549639.413600]: Failure during execution of: Ptp vel_scale: 1.0 acc_scale: 1.0 reference: prbt_base joint goal: [0, 10.0, 0, 0, 0, 0]
[INFO] [1587549639.417148]: Ptp command did fail as expected.
Segmentation fault (core dumped)
```

ct2034, 2020-04-22:
Thanks Guiseppe. I assume you also started a launchfile for the robot and moveit. Could you please also add details on that.

SansoneG, 2020-04-23:
I ran `roslaunch prbt_default_application my_application.launch sim:=false`

Should probably check if this also happens in simulation

martiniil, 2020-04-23:
I did a quick check. I ran:
```
roslaunch prbt_moveit_config moveit_planning_execution.launch sim:=true pipeline:=pilz_command_planner
```
and
```
rosrun pilz_robot_programming demo_program.py
```

The output was exactly the same as above.

martiniil, 2020-07-08:
Found another error which is reported at the end of the script:
```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
```

martiniil, 2020-07-09:
There seems to be a problem with the destruction of the robot object. I created a minimal example:

Approx. 4 failures (segfault) out of 10 runs. No failures if you remove the `__del__` method.
```
#!/usr/bin/env python

from moveit_commander import RobotCommander
from pilz_msgs.msg import MoveGroupSequenceAction

import rospy
from std_srvs.srv import Trigger

PLANNING_GROUP_NAME = "manipulator"
TEST_SERVICE_NAME = "test_service"

class TestClass(object):

    def __init__(self):
        self._service = rospy.Service(TEST_SERVICE_NAME, Trigger, self._callback)
        self._robot_commander = RobotCommander()
        self._robot_commander.get_group(PLANNING_GROUP_NAME).get_current_joint_values()

    def _callback(self, request):
        return True

    def __del__(self):
        self._service.shutdown()

if __name__ == "__main__":
    rospy.init_node('moveit_commander_test_node')

    test = TestClass()
    del(test)

```

martiniil, 2020-07-09:
```
#0  0x0000000000000008 in ?? ()
#1  0x00007ffff364e7fd in ros::Subscription::handleMessage(ros::SerializedMessage const&, bool, bool, boost::shared_ptr<std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > const&, boost::shared_ptr<ros::PublisherLink> const&) () from /opt/ros/melodic/lib/libroscpp.so
#2  0x00007ffff362dcbd in ros::TransportPublisherLink::handleMessage(ros::SerializedMessage const&, bool, bool) () from /opt/ros/melodic/lib/libroscpp.so
#3  0x00007ffff362ddec in ros::TransportPublisherLink::onMessage(boost::shared_ptr<ros::Connection> const&, boost::shared_array<unsigned char> const&, unsigned int, bool) ()
   from /opt/ros/melodic/lib/libroscpp.so
#4  0x00007ffff35af20c in ros::Connection::readTransport() () from /opt/ros/melodic/lib/libroscpp.so
#5  0x00007ffff36288dd in ros::TransportTCP::socketUpdate(int) ()
   from /opt/ros/melodic/lib/libroscpp.so
#6  0x00007ffff365f423 in ros::PollSet::update(int) () from /opt/ros/melodic/lib/libroscpp.so
#7  0x00007ffff35ea355 in ros::PollManager::threadFunc() () from /opt/ros/melodic/lib/libroscpp.so
#8  0x00007ffff2679bcd in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#9  0x00007ffff77cc6db in start_thread (arg=0x7fffcb7fe700) at pthread_create.c:463
#10 0x00007ffff7b05a3f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

martiniil, 2020-07-09:
Conclusion from 2020-07-09:
- report an issue in ros_comm? with above code example. Is the segfault due to faulty destruction during ros shutdown?
- Introduce public `close`/`release` method for `Robot` class and/or recommend using `with`-statements in order to ensure proper destruction of `Robot` objects.

rfeistenauer, 2020-07-09:
It seems like calling `del` on the `RobotCommander` in the `on_shutdown` callback (`rospy.on_shutdown(cb)`) prevents the segfault

agutenkunst, 2020-07-09:
Strange that this service is needed to trigger the issue.

For me it sound like an issue with the Subscriber listening to /joint_states is not properly shut down....

PR #307 (2020-07-13): Deletes robot commander on shutdown to prevent it from segfaulting
Delete RobotController instance before shutting down ROS, to prevent a segfault originating from that class on a normal shutdown.
A Pull Request regarding this problem will be opened on moveit

Closes #263
closed 2020-07-13 by rfeistenauer