PilzDE/pilz_industrial_motion #267: Make master branch buildable under kinetic

ct2034, 2020-04-30:


jschleicher, 2020-05-01:
Does that help? https://github.com/ros-planning/moveit/pull/1893#issuecomment-622144021

ct2034, 2020-05-01:
At the moment I think it is our tests that are failing but it is not possible to get information out of Travis to see why they are failing exactly, just that they are producing no test results. I will try to reproduce this with a kinetic setup

jschleicher, 2020-05-01:
maybe our tests, maybe not: https://github.com/ros-planning/moveit/issues/2042

jschleicher, 2020-05-03:
For instructions to run kinetic in a docker container see https://github.com/ros-planning/moveit/pull/2031#issuecomment-622164655.

Maybe the kinematics solver is missing: prbt_ikfast_manipulator_plugin probably should be added to the build (e.g. via moveit.rosinstall or moveit_resources?), to match the SONAME version of moveit core / master.

jschleicher, 2020-09-28:
kinetic build is no longer relevant.

closed 2020-09-28 by jschleicher