PilzDE/pilz_industrial_motion #289: Add coverage for pytest

agutenkunst, 2020-05-19:
# Code Coverage for pytest with ROS

## Goal
Create coverage for pytest using ROS

## How is this working with unittests?

A python unittest is added to the CMakeLists.txt using
```
catkin_add_nosetests(unittest.py)
```
where the CMake-Macro is defined in [https://github.com/ros/catkin](https://github.com/ros/catkin/blob/dfc93c72d9799f57442d08c2f6e9fd494bba95c1/cmake/test/nosetests.cmake#L59)

here to macro checks for the environment variable `CATKIN_TEST_COVERAGE` and will enable the coverage flags

```
if("$ENV{CATKIN_TEST_COVERAGE}" STREQUAL "1")
  set(_covarg " --with-coverage")
endif()
```
[src](https://github.com/ros/catkin/blob/dfc93c72d9799f57442d08c2f6e9fd494bba95c1/cmake/test/nosetests.cmake#L59-L61)

## How could/should we extend this for pytest?
A python pytest is added to the CMakeLists.txt using
```
add_pytest(pytest.py)
```
where CMake-Macro is define in [https://github.com/machinekoder/ros_pytest](https://github.com/machinekoder/ros_pytest/blob/ba165ad44b854dc6839a9b249ea9b9f1bba5206d/cmake/pytest.cmake#L28)

However inside this Macro the essential part is commented out [see](https://github.com/machinekoder/ros_pytest/blob/ba165ad44b854dc6839a9b249ea9b9f1bba5206d/cmake/pytest.cmake#L51)

### Steps done
- Open PR to rosdistro for python-pytest-cover https://github.com/ros/rosdistro/pull/24929

### Next steps
- [ ] Check if https://github.com/ros/rosdistro/pull/24929 is closed
- [ ] Open PR for https://github.com/machinekoder/ros_pytest for coverage functionality similiar to the catkin CMake-Macro

hslusarek, 2020-05-25:
I opened a draft PR  machinekoder/ros_pytest/pull/6 which should take care of the coverage for the py_test framework.

hslusarek, 2020-05-25:
Currently the [mikeferguson/code_coverage](https://github.com/mikeferguson/code_coverage/blob/master/cmake/Modules/CodeCoverage.cmake#L175) creates a xml file for the python coverage. For the cpp part a html is created, therefore, I suggest to also create a html report instead of a xml report for the python coverage.

hslusarek, 2020-05-25:
Sequence-diagram showing how the of the cmake-macro [`add_code_coverage`](https://github.com/mikeferguson/code_coverage/blob/master/cmake/Modules/CodeCoverage.cmake) works.

[image]

PR #316 (2020-10-01): Use new coverage feature of ros_pytest
Closes #289
closed 2020-10-01 by hslusarek