PilzDE/pilz_industrial_motion #311: Planner plans trajectories that exceed acceleration limit.

SansoneG, 2020-07-23:
### Commit
cb782c272ee6

### Steps to reproduce
How can the issue/bug be reproduced?
1. Switch into AUTO mode
2. Use RViz to plan a movement with velocity- and acceleration scaling 1.0.
3. Execute the movement.

### Expected behavior
Planning and Executing works without error.

### Observed behavior
Planning works, but executing fails. The robot doesn't even start moving.

### Logfiles
```
[ INFO] [1592834842.386682498]: Recovering successful
[ INFO] [1592834848.094568331]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1592834848.094682587]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1592834848.094722985]: Generating PTP trajectory...
[ INFO] [1592834849.925210143]: Execution request received
[ERROR] [1592834849.972380926]: Acceleration limit violated by joint prbt_joint_6. Desired acceleration: 4.43134rad/s^2, limit:  4rad/s^2.
[ INFO] [1592834849.972646862]: Controller prbt/manipulator_joint_trajectory_controller/ successfully finished
[ WARN] [1592834849.972706115]: Controller handle prbt/manipulator_joint_trajectory_controller/ reports status PREEMPTED
[ INFO] [1592834849.972743617]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1592834849.972813891]: Execution completed: PREEMPTED
[ INFO] [1592834850.003672288]: PREEMPTED: Preempted
```
### Reference to test
https://github.com/PilzDE/pilz_robots/blob/melodic-devel/pilz_control/test/acceptance_test_acceleration_limit.md

martiniil, 2020-07-23:
LIN trajectory with acceleration scale `0.1`:
[image]

martiniil, 2020-07-23:
The oscillations shown above happen for LIN and CIRC.

I guess the crucial point is the computation of the velocities in the trajectory points. For PTP the velocities are obtained directly from the trapezoidal profile. Opposed to that, LIN and CIRC are planned in cartesian space, the positions are transformed via the inverse kinematic and the velocities are computed as follows:
https://github.com/PilzDE/pilz_industrial_motion/blob/dd9b30aaaab9c3cdb4b524d4970dfeb91e48e62f/pilz_trajectory_generation/src/trajectory_functions.cpp#L290
Using these approximated values, the positions and velocities do not "fit together" very well which leads to the oscillations.

fmros, 2021-06-08:
> d values, the positions and velocities do not "fit together" very well which leads to the oscillations.

I am seeing similar oscillations of the velocity profile from a LIN planned trajectory. But evaluating the output-trajectories with the rqt_joint_trajectory_plot tool shows me, that the trajectories are clean before entering the joint_trajectory_controller.

After some investigation, i found [this](https://github.com/ros-planning/moveit/blob/cdda16fdee21f05d6a6209d59e08cc6086d9efb6/moveit_planners/pilz_industrial_motion_planner/include/pilz_industrial_motion_planner/trajectory_generator.h#L109) parameter. in the trajectory_generator.h. I increased the frequency by a order of 10, so sampling_time = 0.01 and the resulting velocity/acceleration trajectories become free of oscillations. I assume that the sampling_time of 10Hz is to low for a smooth interpolation of the joint-trajectory controller?

Unfortunately I did not find yet, where this parameter could be passed to the trajectory_generator_*, so it remains hard-coded for me at the moment.

Would be happy to get a feedback to this thoughts

closed 2023-01-27 by martiniil