PilzDE/pilz_industrial_motion #319: SequenceMoveGroup: find out which source command the trajectory output belongs to

machinekoder, 2020-11-18:
**Is your feature request related to a problem? Please describe.**
In our particular use case of the sequence planner, we use it to combine multiple commands in a robot program. It would be nice to give the user feedback on which source command the current execution state belongs to. For this purpose, it would be required to map the current time_from_start to a particular source command.

**Describe the solution you'd like**
The MoveGroupSequenceResult allows multiple planned trajectory results. Splitting up the trajectory into multiple parts might be an option to mark the individual source commands the trajectory was generated from.

**Describe alternatives you've considered**
The MoveGroupSequenceResult could be extended to contain start and end indizes to indicate which source command the segments of the trajectory belong to.

**Additional context**
none

machinekoder, 2020-11-18:
Note that the vanilla `position_trajectory_controller` doesn't give feedback on the position in the trajectory either. However, we are already working on extending this for pause/resume support. Given that we have the feedback working as wished, feedback on the generated segments would be meaningful.

jschleicher, 2021-01-08:
Yep, we would be glad to see such an addition as PR. This is related to #214 . 
Please note, that the code is now moved as plugin into [upstream moveit](https://github.com/ros-planning/moveit/pull/1893), so please propose PRs over there.

closed 2023-01-27 by martiniil