PilzDE/pilz_industrial_motion #325: **Build Error** : ‘static bool planning_scene::PlanningScene::isEmpty(const PlanningScene&)’ is deprecated: Use moveit/utils/message_checks.h instead [-Werror=deprecated-declarations]

srujan-aris, 2020-12-09:
### Commit
At which commit was the issue/bug observed?
running ros-noetic on Ubuntu 20.04
building the melodic-devel

### Steps to reproduce
How can the issue/bug be reproduced?
```
mkdir -p pilz_ws/src
cd pilz_ws/src
git clone https://github.com/PilzDE/pilz_common.git
git clone https://github.com/PilzDE/pilz_industrial_motion.git
catkin build
```
### Expected behavior
build 100%

### Observed behavior

### Logfiles

```
srujan@srujan-aris1:~/pilz_ws$ catkin build
---------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/noetic
Workspace:                   /home/srujan/pilz_ws
---------------------------------------------------------
Build Space:        [exists] /home/srujan/pilz_ws/build
Devel Space:        [exists] /home/srujan/pilz_ws/devel
Install Space:      [unused] /home/srujan/pilz_ws/install
Log Space:         [missing] /home/srujan/pilz_ws/logs
Source Space:       [exists] /home/srujan/pilz_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
---------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                                                                                                           
[build] Updating package table.                                                                                                                                      
Starting  >>> catkin_tools_prebuild                                                                                                                                  
Finished  <<< catkin_tools_prebuild                           [ 1.3 seconds ]                                                                                        
Starting  >>> pilz_extensions                                                                                                                                        
Starting  >>> pilz_msgs                                                                                                                                              
Starting  >>> pilz_store_positions                                                                                                                                   
Starting  >>> pilz_utils                                                                                                                                             
Finished  <<< pilz_utils                                      [ 1.9 seconds ]                                                                                        
Starting  >>> pilz_testutils                                                                                                                                         
Finished  <<< pilz_extensions                                 [ 1.9 seconds ]                                                                                        
Finished  <<< pilz_store_positions                            [ 2.4 seconds ]                                                                                        
Finished  <<< pilz_msgs                                       [ 4.5 seconds ]                                                                                        
Starting  >>> pilz_industrial_motion_testutils                                                                                                                       
Finished  <<< pilz_testutils                                  [ 4.4 seconds ]                                                                                        
Finished  <<< pilz_industrial_motion_testutils                [ 10.2 seconds ]                                                                                       
Starting  >>> pilz_trajectory_generation                                                                                                                             
_____________________________________________________________________________________________________________________________________________________________________
Errors     << pilz_trajectory_generation:make /home/srujan/pilz_ws/logs/pilz_trajectory_generation/build.make.000.log                                                
[... 151 lines cut ...]
                 from /home/srujan/pilz_ws/src/pilz_industrial_motion/pilz_trajectory_generation/src/move_group_sequence_action.cpp:39:
/home/srujan/source_moveit/src/moveit/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h:944:75: note: declared here
   [[deprecated("Use moveit/utils/message_checks.h instead")]] static bool isEmpty(const moveit_msgs::PlanningScene& msg);
                                                                           ^~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/sequence_capability.dir/build.make:63: CMakeFiles/sequence_capability.dir/src/move_group_sequence_action.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:3754: CMakeFiles/sequence_capability.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
cd /home/srujan/pilz_ws/build/pilz_trajectory_generation; catkin build --get-env pilz_trajectory_generation | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
.....................................................................................................................................................................
Failed     << pilz_trajectory_generation:make                 [ Exited with code 2 ]                                                                                 
Failed    <<< pilz_trajectory_generation                      [ 1 minute and 15.0 seconds ]                                                                          
Abandoned <<< pilz_industrial_motion                          [ Unrelated job failed ]                                                                               
Abandoned <<< pilz_robot_programming                          [ Unrelated job failed ]                                                                               
[build] Summary: 7 of 10 packages succeeded.                                                                                                                         
[build]   Ignored:   None.                                                                                                                                           
[build]   Warnings:  None.                                                                                                                                           
[build]   Abandoned: 2 packages were abandoned.                                                                                                                      
[build]   Failed:    1 packages failed.                                                                                                                              
[build] Runtime: 1 minute and 44.5 seconds total.                                                                                                                    
[build] Note: Workspace packages have changed, please re-source setup files to use them.
srujan@srujan-aris1:~/pilz_ws$ 
```

srujan-aris, 2020-12-09:
I tried underlaying a workspace which has moveit (melodic-devel) installed from source, and the **build was successful**. Is there a way to do this for noetic? Please suggest

martiniil, 2021-01-18:
Thanks for reporting this issue. What is your intention? Would you like to use the industrial planner in noetic? Then I suggest that you simply remove `pilz_industrial_motion` since the planner was merged into `moveit` (branch: `master`).

Feel free to re-open or pose another issue if your problem was not solved.

srujan-aris, 2021-01-19:
That's a good news @martiniil. 
My intention is to generate Move_J (PTP) and Move_L (LIN) type of trajectories and perform collision checking for a UR10. 
Previously I had no luck using this planner with moveit-noetic. I had to install moveit-melodic from source to use this planner. 
I will try once again to use this planner with moveit-noetic and see if it works.

jschleicher, 2021-01-21:
@srujan-aris Yep, our planner was integrated upstream as [pilz_industrial_motion_planner](https://moveit.ros.org/moveit/pilz/motion%20planner/2020/12/17/Pilz-Plugin-for-MoveIt.html).

srujan-aris, 2021-02-03:
@jschleicher , I see the pilz_industrial_motion_planner upstream. I have one quick question, does moveit_setup_assistant suppose to generate config files related to pilz_industrial_motion_planner in the config package? I couldn't see any of those generated.

jschleicher, 2021-02-04:
Yes, the moveit_setup_assistant does generate the files - you have to just select `pipeline:=pilz_industrial_motion_planner` to use them. The current moveit master isn't released as binary, so you still need to build moveit from source.

closed 2021-02-09 by srujan-aris