PilzDE/pilz_industrial_motion #340: Running demo_program.py

robertbeach, 2021-02-23:
### Commit
Noetic

### Steps to reproduce
Launch demo.launch
try to run demo_program.py  

### Expected behavior
It should compile without errors

### Observed behavior
When I run demo_program.py I get an error message during the python "compile" phase indicating an item cannot be found.

@bob-u20:~/catkin_ws_otrajobj/src/kinova-ros/kinova_demo/nodes/kinova_demo$ python demo_program.py
Traceback (most recent call last):
  File "demo_program.py", line 18, in <module>
    from pilz_robot_programming import *
  File "/home/bob/catkin_ws_otrajobj/devel/lib/python3/dist-packages/pilz_robot_programming/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 23, in <module>
  File "/home/bob/catkin_ws_otrajobj/src/pilz_robot_programming/src/pilz_robot_programming/commands.py", line 25, in <module>
    from pilz_msgs.msg import MoveGroupSequenceGoal, MotionSequenceItem
ImportError: cannot import name 'MoveGroupSequenceGoal' from 'pilz_msgs.msg' (/home/bob/catkin_ws_otrajobj/devel/lib/python3/dist-packages/pilz_msgs/msg/__init__.py)

----
A similar program occurs in robot.py. 

The problem seems to be that the MoveGroupSequenceGoal msg has been moved from pilz_common to moveit_msgs. The routines command.py and robot.py need to be modified accordingly.

jschleicher, 2021-02-24:
@robertbeach Thank you for reporting your issue. We have currently a branch https://github.com/PilzDE/pilz_industrial_motion/pull/331 that you could use with a moveit source build. We will merge it as soon as the binary release of moveit is out.
Please reopen, if you experience any problems on that branch.

closed 2021-02-24 by jschleicher

robertbeach, 2021-02-24:
Hi

Thanks for the response. I thought the binary release of Noetic is already released. Is there something in the latest moveit source build that your software depends upon?

Branch 331 only refers to the pilz_industrial_motion directory. What about Pilz_robots and pilz_common? There is a noetic tagged branch for them, should I use that one?

Thanks

Bob

[External Email]

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

robertbeach, 2021-02-25:
Hello again

I cloned #331 as well as pliz_common (noetic) into my catkin_ws directory. When I try to build I get errors such as:

/home/bob/pilz_catkin_ws/src/pilz_trajectory_generation/include/pilz_trajectory_generation/command_list_manager.h:28:10: fatal error: pilz_msgs/MotionSequenceRequest.h: No such file or directory
   28 | #include "pilz_msgs/MotionSequenceRequest.h"

Is the noetic build usable?

?Does the melodic build run on noetic?

Thanks for your help.

martiniil, 2021-02-25:
Hello @robertbeach.
Our command planner has been moved to moveit, but we are still waiting for a release of it. Unfortunately, using it on ROS noetic is not that straightforward at the moment. Maybe consider switching to melodic until everything is released. If you still want to build on ROS noetic, you need the following:

- moveit from source, see https://moveit.ros.org/install/source/, https://github.com/ros-planning/moveit/blob/master/moveit.rosinstall
- In another workspace: pilz_robots & pilz_industrial_motion from source, branch: noetic-devel
- Install all required dependencies, then build moveit, source it and you should be able to build our packages and execute them.

robertbeach, 2021-02-26:
Hello

I have switched over to melodic on ubuntu 18.04. The build process went better but there are still problems.

The first thing I want to do is to run the pilz software in Rviz using a simulated robot. However the build sequence tries  to build a system for an actual physical robot using CAN and Modbus interfaces. While I was able to find a canbus package, there are too many modbus options to consider, especially because I don’t have a physical pilz robot.  Can you suggest how I can build an Rviz only system?

The second thing I want to do is to run the software on a different (non Pilz) physical robot. It is not clear whether your software uses the regular trajectory controller that other moveit planners use or whether some other controller is required. Could you clarify this?

Thanks

Bob

[External Email]

jschleicher, 2021-02-26:
@robertbeach Did you follow the tutorial steps on [ROS wiki](http://wiki.ros.org/pilz_robots/Tutorials/MoveRobotWithPilzCommand_planner)? If you don't have a physical safety controller, you can use a `fake_safety_interface` as in our [templates](https://github.com/PilzDE/pilz_application_templates/blob/65a89ad61fb924f1b139bf02a4d7cfcd4fd0c6e1/prbt_default_application/launch/my_application.launch#L111).

robertbeach, 2021-04-28:
Hello

What is the state of pilz on noetic? Is it part of the binary release or is it still build from source only?

Thanks

Bob

martiniil, 2021-04-29:
@robertbeach Not yet, but we are almost there.

robertbeach, 2021-04-30:
Is there a source build that works?

[External Email]

martiniil, 2021-05-03:
Yes, I described it here: https://github.com/PilzDE/pilz_industrial_motion/issues/340#issuecomment-785722850

martiniil, 2021-06-07:
fyi: the binary release is out. The planner was renamed to `pilz_industrial_motion_planner`.