PilzDE/pilz_industrial_motion #345: Simple replacement for compute_cartesian_path (python)

MatMcT, 2021-07-09:
Hi guys,

After a few days of searching, I couldn't seem to find a solution to what I was looking for. 

I was looking to use the functionality of MotionSequenceRequest as a replacement for MoveGroupCommander.compute_cartesian_path. 

I reading the documentation [here](https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html)

I tried using the functions (PTP, LIN etc) from [demo_program.py](https://github.com/PilzDE/pilz_industrial_motion/blob/melodic-devel/pilz_robot_programming/examples/demo_program.py) but couldn't seem to get past:
`Waiting for connection to service /prbt/get_speed_override...`
From what I gather, it's looking for a real PRBT connected? I should disclose I'm using a UR5e. 

After hours of digging and frustration, I just wrote my own wrapper. It uses MoveGroupCommander as a parent class so you have access to all the underlying methods. By using MotionSequenceRequest it solves a few shortcomings that compute_cartesian_path has ie. no speed scaling and provides the benefit of using a blend radius. 

Does something like already exist? If not I'd be happy to share.

o0stinger0o, 2021-07-25:
Hi MatMcT,
Recently, I try to find a method to control the speed of cartesian path.  
My work environment is  ROS-melodic,moveit,UR10e,and webots.
I've tried Modify the **joint_limits.yaml**  and use  [retime_trajectory()](http://docs.ros.org/en/melodic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a5fb1d772b6d248a6c4113412eacb4330),[set_max_acceleration_scaling_factor(), set_max_velocity_scaling_factor()](http://docs.ros.org/en/melodic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a0c46ebb5a29bbfe84882f5f296352b90)
but the speed and accelerate depends on the webots simulation speed.
So i want to try Pilz Industrial Motion Planner,but the tutorials make me confused.

could you share your results? Thank you very much.

jschleicher, 2021-08-06:
For speed_override without the corresponding safety controller you could simply comment the lines in robot.py https://github.com/PilzDE/pilz_industrial_motion/blob/5a0efb8efc287db659ff3a3075549243b381725c/pilz_robot_programming/src/pilz_robot_programming/robot.py#L542-L546 or use the fake_speed_override as in several test cases: https://github.com/PilzDE/pilz_industrial_motion/blob/d3b255df6f35da69af0284961f2a4e0a7aecd85b/pilz_robot_programming/test/integrationtests/integrationtest_self_collision.test#L41

closed 2021-08-06 by jschleicher