PilzDE/pilz_industrial_motion #348: Cartesian limits are not fully set for LIN trajectory generator

s265452, 2021-07-28:
Hi everyone,
I'm using MoveIt in order to control my Jaco2 7 DoF.
I followed MoveIt tutorial:

https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#pilz-industrial-motion-planner

After that, I can Plan & Execute a trajectory with Pilz Industrial Motion Planner only in PTP mode.

When I try to Plan & Execute a trajectory either LIN or CIRC I have an error like:

 Cartesian limits not set for LIN trajectory generator.
[ERROR] [1627481556.196751856]: Exception caught: 'Cartesian limits are not fully set for LIN trajectory generator.'

Can anyone help me out with this?

martiniil, 2021-07-28:
The error indicates that for some reason the planner could not read the cartesian limits. Make sure you have set all the limits as described in the moveit tutorial. The following example with another robot shows how the cartesian limits are loaded correctly onto the parameter server:
https://github.com/PilzDE/pilz_robots/blob/noetic-devel/prbt_moveit_config/launch/planning_context.launch

s265452, 2021-07-29:
Thank you so much! 
Your tip solved the problem.

closed 2021-07-29 by s265452