PilzDE/pilz_industrial_motion #349: Pilz robot programming CIRC seems to be broken

theofficialgman, 2021-08-10:
running ros-noetic (may be an issue on other version as well)

I have observed that the circle command in pilz robot programming produces errors in the output and doesn't not follow an arc.
I've tested with my own simple code as well as the demo program launched as below:
```
roslaunch moveit_resources_prbt_moveit_config demo.launch
rosrun pilz_robot_programming demo_program.py
```
when CIRC is called there are errors such as this in the output and a circular path is NOT followed. this is running the demo program and the demo launch file
```
[ INFO] [1628606389.465833816]: Planning attempt 1 of at most 1
[ WARN] [1628606389.466642623]: Cannot find planning configuration for group 'manipulator' using planner 'CIRC'. Will use defaults instead.
[ INFO] [1628606389.472730080]: Using link_prefix: 'prbt_'
[ INFO] [1628606389.475333190]: Planner configuration 'manipulator' will use planner 'geometric::RRT'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1628606389.475952396]: manipulator: Allocating specialized state sampler for state space
[ INFO] [1628606389.476167247]: manipulator/manipulator: Starting planning with 1 states already in datastructure
[ERROR] [1628606389.476334130]: Unable to sample a point inside the constraint region
[ERROR] [1628606389.476406266]: Unable to sample a point inside the constraint region
[ERROR] [1628606389.476502660]: Unable to sample a point inside the constraint region
[ERROR] [1628606389.482357902]: Unable to sample a point inside the constraint region
[ERROR] [1628606389.482481433]: Unable to sample a point inside the constraint region
[ERROR] [1628606389.482567240]: Unable to sample a point inside the constraint region
```

On my own robot I have also tried a simple circle around the middle origin which fails and ends up just doing a straight line
```
  r.move(Ptp(goal=Pose(position=Point(0.0,0.5,0.495500000), orientation=Quaternion(*transformations.quaternion_from_euler(0.0,0.0,0.0))), acc_scale=0.5, vel_scale=1, reference_frame='world', target_link='tool0', planning_group='manipulator'))
  r.move(Circ(goal=Pose(position=Point(0.5000,0.00,0.49550000), orientation=Quaternion(*transformations.quaternion_from_euler(0.0,0.0,0.0))), center=Point(0.0,0.0,0.49550000), acc_scale=0.5, vel_scale=0.24, reference_frame='world', target_link='tool0', planning_group='manipulator'))
```

jschleicher, 2021-08-13:
Probably you haven't loaded the Pilz planner at all but are using an OMPL configuration, that ignores the constraint:
> [ WARN] [1628606389.466642623]: Cannot find planning configuration for group 'manipulator' using planner 'CIRC'. Will use defaults instead.

Please also note, that the planner was integrated into upstream MoveIt in the meantime, you can find additional information in their Tutorials: https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html

Please use answers.ros.org for further questions, how to setup MoveIt correctly.

theofficialgman, 2021-08-18:
@jschleicher probably you should have read my message
I have tested using both the moveit_resources_prbt_moveit_config demo.launch (ie: your own launch file) as well as my own launch file with pilz selected. both don't work, the pilz planner is loaded and in my own program I've selected it as the chosen planner.

its the fault of the pilz robot programming api, I can do circle moves if I correctly generate the move request myself

jschleicher, 2021-08-19:
If there is a bug in pilz_robot_programming python scripts, feel free to file a pull request. Sorry for my misunderstanding!

martiniil, 2021-08-23:
The log output you gave does not support that this is an error in the pilz_robot_programming api.

If you use [demo.launch](https://github.com/ros-planning/moveit_resources/blob/master/prbt_moveit_config/launch/demo.launch) from moveit_resources, you have to specify `pipeline:=pilz_industrial_motion_planner`.

ct2034, 2021-12-15:
@theofficialgman I am closing this because it is not reproducible with the correct pipeline specified (i.e. `roslaunch prbt_moveit_config demo.launch pipeline:=pilz_industrial_motion_planner`)
see default param in https://github.com/PilzDE/pilz_robots/blob/noetic-devel/prbt_moveit_config/launch/demo.launch

If the problem still persist for you, please post the output that you get with the command above.

closed 2021-12-15 by ct2034