PilzDE/pilz_industrial_motion #357: How to use pills robot programming with Staubli robots ?

arvind1455, 2022-04-19:
I have been trying to use pilz robot programming with Staubli robots and failed eventually. So wanted to know what are the steps to use the code with Staubli robots

ct2034, 2022-06-24:
Hi Kamesh. Can you please give me more detailed info on what you are trying to do. Please tell me which version you are using and send me the launch files you are using so that I can reproduce your errors. 
We can not provide tutorials for every type of robot but if your robot works with moveit, it should also work with our planner.

rijalmadhav12, 2022-09-27:
for example panda robot which is used in moveit demo.

rijalmadhav12, 2022-09-27:
i want to use pilz robot programming to sequence motion of panda arm that is used in moveit demo. What are the steps I need to follow. please guide me.

closed 2023-01-27 by martiniil